raspberrypi/utils #3: truncated output

arachnaut, 2019-08-29:
The line

"cat /sys/firmware/devicetree/base/model"

for Raspian Pi 4 puts out some garbage characters that seem to truncate the tee output 

Here is what I see when I comment that line away:

System Information
------------------

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage5

Linux raspberrypi 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l GNU/Linux
Hardware	: BCM2835
Revision	: c03111
...

Otherwise it is truncated at the start.

pelwell, 2019-09-01:
raspi-info works for me, and the content of the model property looks correct:
```
pi@raspberrypi:~ $ od -Ax -tx1 /sys/firmware/devicetree/base/model 
000000 52 61 73 70 62 65 72 72 79 20 50 69 20 33 20 4d
000010 6f 64 65 6c 20 42 20 50 6c 75 73 20 52 65 76 20
000020 31 2e 33 00
000024
```
Please run the same command and show the output.

1stcall, 2022-09-08:
I notice if I open my output from raspinfo in vscode, the model of the pi line is terminated with a (null).
System Information
------------------

Raspberry Pi 4 Model B Rev 1.4
and 
System Information
------------------

Raspberry Pi Compute Module 4 Rev 1.0

[raspinfo.txt](https://github.com/raspberrypi/utils/files/9525952/raspinfo.txt)

pelwell, 2022-09-08:
Indeed - that is how strings are represented in Device Tree. raspinfo could be modified to surpress the NUL, but it's not been a problem so far (you are the first to comment).

closed 2022-09-08 by pelwell