raspberrypi/utils #17: Please provide packaging for vclog

hailfinger, 2023-01-09:
On 64-bit Raspberry Pi OS, there are three ways to get the VC logs:
- statically linked vcdbg from some Google Drive
- dynamically linked vcdbg from the armhf repo, needs additional armhf libraries to work
- compile vclog yourself

It would be great to have vclog available as Debian package in the Raspberry Pi OS repositories.
Thank you!

XECDesign, 2023-02-14:
You should be able to install `raspi-utils` to get `vclog` now.

hailfinger, 2023-02-17:
Yes, thank you. However, it seems that vclog doesn't work with older kernels:
```
root@pi-test-1:# vclog -a
Could not read from Device Tree log starts and log size
root@pi-test-1:# vclog -m
Could not read from Device Tree log starts and log size

raspberrypi-bootloader 1:1.20220331-1
raspberrypi-kernel 1:1.20220331-1
root@pi-test-1:# vcgencmd bootloader_version
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1626781199
capabilities 0x0000001f
root@pi-test-1:# uname -a
Linux pi-test-1 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

hailfinger, 2023-02-17:
It does work perfectly with the current kernel:

```
root@pi-test-1:~# vclog -a
No messages available

raspberrypi-kernel 1:1.20230106-1
raspberrypi-bootloader 1:1.20230106-1
root@pi-test-1:~# uname -a
Linux pi-test-1 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux
```

Would it make sense to add a dependency on a minimum kernel version to raspi-utils?

pelwell, 2023-02-17:
It won't be the kernel version - try updating the firmware.

lurch, 2023-10-12:
This can probably be closed now, as `vclog` is included by default in both 32-bit and 64-bit versions of Raspberry Pi OS Bookworm.

closed 2023-10-13 by pelwell