raspberrypi/utils #21: vclog debug build with -Og fails with maybe-uninitialized errors

HiassofT, 2023-02-24:
Creating a debug build of vclog fails with maybe-uninitialized errors.

eg on RPi4 running current RPiOS bullseye lite:
```
pi@raspberrypi:~/install/raspi-utils/vclog/build $ cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_C_FLAGS="-Og" ..
...
pi@raspberrypi:~/install/raspi-utils/vclog/build $ make VERBOSE=1
...
/usr/bin/cc   -Og -Wall -Wextra -Werror -pedantic -g -o CMakeFiles/vclog.dir/vcdbg.c.o -c /home/pi/install/raspi-utils/vclog/vcdbg.c
/home/pi/install/raspi-utils/vclog/vcdbg.c: In function ‘find_hdr_requested_log’:
/home/pi/install/raspi-utils/vclog/vcdbg.c:576:10: error: ‘val’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  576 |   return correct_to_be_in_virtual_loc(toc, val);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/install/raspi-utils/vclog/vcdbg.c: In function ‘main’:
/home/pi/install/raspi-utils/vclog/vcdbg.c:345:37: error: ‘header_current_msg’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  345 |         size_copying = end_all_msgs - (char *)header_current_msg;
      |                        ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

These seem to be genuine issues in the code, `find_hdr_requested_log`, doesn't handle the log_unknown enum (and leaves val unset) and from a quick glance there might be (potential) cases in main that leave find_hdr_requested_log unset

popcornmix, 2023-02-24:
Should be fixed by https://github.com/raspberrypi/utils/pull/22

HiassofT, 2023-02-24:
thanks a lot for the quick fix!

closed 2023-02-24 by HiassofT