raspberrypi/utils #23: Integer overflow in hexadecimal number at /usr/bin/ovmerge line 1888

macmpi, 2023-03-11:
```
# ovmerge
Integer overflow in hexadecimal number at /usr/bin/ovmerge line 1888.
Usage: ovmerge <options> <ovspec>
  where <ovspec> is the name of an overlay, optionally followed by
    a comma-separated list of parameters, each with optional '=<value>'
    assignments. The presence of any parameters, or a comma followed by
    no parameters, removes the parameter declarations from the merged
    overlay to avoid a potential name clash.
  and <options> are any of:
    -b <branch>  Read files from specified git branch
    -c      Include 'redo' comment with command line (c.f. '-r')
    -e      Expand mode - list non-skipped lines in order of inclusion
    -h      Display this help info
    -i      Show include hierarchy for each file
    -l      Like expand mode, but labels each line with source file
    -n      No .dts file header (just parsing .dtsi files)
    -p      Emulate Pi firmware manipulation
    -r      Redo command comment in named files (c.f. '-c')
    -s      Sort nodes and properties (for easy comparison)
    -t      Trace
    -w      Show warnings
```

compiled with Alpine/musl in case that matters...

pelwell, 2023-03-11:
It's a Perl script so there is no compilation. I'm guessing you are running it on a Pi in a 32-bit userland (something I've never tried)? It's an annoying result, but should be fixable with some effort.

macmpi, 2023-03-11:
>It's a Perl script so there is no compilation. 

Sure but compiled whole project which also contains compiled binaries: did not check if any is called from this script.
Yes is on 32-bit PiZeroW.

pelwell, 2023-03-13:
Which Perl does your OS use? 32-bit RPiOS Perl is quite happy with 64-bit integer values, although I note that using a parameter to overwrite a pair of cells with a 64-bit value puts the potentially large value in the first cell and doesn't touch the second.

macmpi, 2023-03-14:
Thanks for feedback.
Alpine 3.17 used in this test ships [perl 5.36.0](https://git.alpinelinux.org/aports/tree/main/perl/APKBUILD?h=3.17-stable).

pelwell, 2023-03-14:
```perl
use Config qw(myconfig);
print myconfig();
```
shows:
```
Summary of my perl5 (revision 5 version 32 subversion 1) configuration:
    osname=linux
    osvers=4.19.0
    archname=arm-linux-gnueabihf-thread-multi-64int
    uname='linux localhost 4.19.0 #1 smp debian 4.19.0 armv7l gnulinux '
    config_args='-Dmksymlinks -Dusethreads -Duselargefiles -Dcc=arm-linux-gnueabihf-gcc -Dcpp=arm-linux-gnueabihf-cpp -Dld=arm-linux-gnueabihf-gcc -Dccflags=-DDEBIAN -Wdate-time -D_FORTIFY_SOURCE=2 -g -O2 -ffile-prefix-map=/dummy/build/dir=. -fstack-protector-strong -Wformat -Werror=format-security -Dldflags= -Wl,-z,relro -Dlddlflags=-shared -Wl,-z,relro -Dcccdlflags=-fPIC -Darchname=arm-linux-gnueabihf -Dprefix=/usr -Dprivlib=/usr/share/perl/5.32 -Darchlib=/usr/lib/arm-linux-gnueabihf/perl/5.32 -Dvendorprefix=/usr -Dvendorlib=/usr/share/perl5 -Dvendorarch=/usr/lib/arm-linux-gnueabihf/perl5/5.32 -Dsiteprefix=/usr/local -Dsitelib=/usr/local/share/perl/5.32.1 -Dsitearch=/usr/local/lib/arm-linux-gnueabihf/perl/5.32.1 -Dman1dir=/usr/share/man/man1 -Dman3dir=/usr/share/man/man3 -Dsiteman1dir=/usr/local/man/man1 -Dsiteman3dir=/usr/local/man/man3 -Duse64bitint -Dman1ext=1 -Dman3ext=3perl -Dpager=/usr/bin/sensible-pager -Uafs -Ud_csh -Ud_ualarm -Uusesfio -Uusenm -Ui_libutil -Ui_xlocale -Uversiononly -DDEBUGGING=-g -Doptimize=-O2 -dEs -Duseshrplib -Dlibperl=libperl.so.5.32.1'
```
Notice the `-Duse64bitint`, which I'm assuming your build lacks.
I'm not planning to "fix" this soon, but I'll take a PR if you want. In the meantime, if you delete the `8=>0xffffffffffffffff` in the `%masks` hash on line 1888 you'll probably find it work for all overlays except ramoops.

macmpi, 2023-03-14:
>Notice the -Duse64bitint, which I'm assuming your build lacks.

Is `undef` indeed: thanks.
Won't be able to craft a PR for that I'm afraid, and will leave as-is now this is well documented: thanks.
I'm not sure many user would use _ovmerge_ in such setup anyway...

My intend is essentially to shape-up an Alpine package with _vclog_/_raspinfo_ to provide base tools to help users debug running instances with help of the Pi community at large: I just added other utilities for sake of completeness, though they are more development related...
Thanks again for your great support.

```
Summary of my perl5 (revision 5 version 36 subversion 0) configuration:
   
  Platform:
    osname=linux
    osvers=5.15.32-0-lts
    archname=armv8l-linux-thread-multi
    uname='linux build-3-17-armhf 5.15.32-0-lts #1-alpine smp mon, 28 mar 2022 13:09:00 +0000 armv8l linux '
    config_args='-des -Dcccdlflags=-fPIC -Dcccdlflags=-fPIC -Dccdlflags=-rdynamic -Dprefix=/usr -Dprivlib=/usr/share/perl5/core_perl -Darchlib=/usr/lib/perl5/core_perl -Dvendorprefix=/usr -Dvendorlib=/usr/share/perl5/vendor_perl -Dvendorarch=/usr/lib/perl5/vendor_perl -Dsiteprefix=/usr/local -Dsitelib=/usr/local/share/perl5/site_perl -Dsitearch=/usr/local/lib/perl5/site_perl -Dlocincpth=  -Doptimize=-Os -fomit-frame-pointer -Duselargefiles -Dusethreads -Duseshrplib -Dd_semctl_semun -Dman1dir=/usr/share/man/man1 -Dman3dir=/usr/share/man/man3 -Dinstallman1dir=/usr/share/man/man1 -Dinstallman3dir=/usr/share/man/man3 -Dman1ext=1 -Dman3ext=3pm -Dcf_by=Alpine -Ud_csh -Dusenm'
    hint=recommended
    useposix=true
    d_sigaction=define
    useithreads=define
    usemultiplicity=define
    use64bitint=undef
    use64bitall=undef
    uselongdouble=undef
    usemymalloc=n
    default_inc_excludes_dot=define
  Compiler:
    cc='cc'
    ccflags ='-D_REENTRANT -D_GNU_SOURCE -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64'
    optimize='-Os -fomit-frame-pointer'
    cppflags='-D_REENTRANT -D_GNU_SOURCE -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong'
    ccversion=''
    gccversion='12.2.1 20220924'
    gccosandvers=''
    intsize=4
    longsize=4
    ptrsize=4
    doublesize=8
    byteorder=1234
    doublekind=3
    d_longlong=define
    longlongsize=8
    d_longdbl=define
    longdblsize=8
    longdblkind=0
    ivtype='long'
    ivsize=4
    nvtype='double'
    nvsize=8
    Off_t='off_t'
    lseeksize=8
    alignbytes=8
    prototype=define
  Linker and Libraries:
    ld='cc'
    ldflags =' -fstack-protector-strong -L/usr/local/lib'
    libpth=/usr/local/lib /usr/lib /lib
    libs=-lpthread -ldl -lm -lcrypt -lutil -lc
    perllibs=-lpthread -ldl -lm -lcrypt -lutil -lc
    libc=/usr/lib/libc.a
    so=so
    useshrplib=true
    libperl=libperl.so
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_dlopen.xs
    dlext=so
    d_dlsymun=undef
    ccdlflags='-rdynamic -Wl,-rpath,/usr/lib/perl5/core_perl/CORE'
    cccdlflags='-fPIC'
    lddlflags='-shared -Os -fomit-frame-pointer -L/usr/local/lib -fstack-protector-strong'
```

pelwell, 2023-03-14:
There's a workaround in #26 that should work for most situations.

macmpi, 2023-03-14:
works in my case :+1:

macmpi, 2023-03-14:
Thanks, closing then.

closed 2023-03-14 by macmpi