raspberrypi/utils #29: vclog doesn't seem to work

vfazio, 2023-08-03:
I can't get meaningful logs displayed. This is in direct contrast to the old vcdbg.

I have the following deployment:

```
root@rpi-87fa00:~# vcgencmd version
Apr 25 2023 18:33:04
Copyright (c) 2012 Broadcom
version d7f9c2b4ef7e4a8c0b04374a879ce89d7a948453 (clean) (release) (start_x)

root@rpi-87fa00:~# uname -a
Linux rpi-87fa00 6.1.42-v8 #1 SMP PREEMPT Wed Aug  2 20:49:45 UTC 2023 aarch64 GNU/Linux

root@rpi-87fa00:~# /lib/aarch64-linux-gnu/libc.so.6
GNU C Library (Debian GLIBC 2.36-9+deb12u1) stable release version 2.36.
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 12.2.0.
libc ABIs: UNIQUE ABSOLUTE
Minimum supported kernel: 3.7.0
For bug reporting instructions, please see:
<http://www.debian.org/Bugs/>.
```

I've built vclog off of 8fc892d82f7c58e2adc58354cf62a73d1b6c6850

```
root@rpi-87fa00:/var/tmp/tmp.MoicJspBpb/utils/build/# git rev-parse HEAD
8fc892d82f7c58e2adc58354cf62a73d1b6c6850

root@rpi-87fa00:/var/tmp/tmp.MoicJspBpb/utils/build# cmake ..
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /var/tmp/tmp.MoicJspBpb/utils/build
root@rpi-87fa00:/var/tmp/tmp.MoicJspBpb/utils/build# make vclog
[ 50%] Building C object vclog/CMakeFiles/vclog.dir/vcdbg.c.o
[100%] Linking C executable vclog
[100%] Built target vclog
```

but upon execution, i see nothing
```
root@rpi-87fa00:/var/tmp/tmp.MoicJspBpb/utils/build# ./vclog/vclog -m
root@rpi-87fa00:/var/tmp/tmp.MoicJspBpb/utils/build# ./vclog/vclog -a

```

when i run the current 64bit statically compiled vcdbg i do get output:

```
root@rpi-87fa00:/var/tmp/tmp.MoicJspBpb/utils/build# ~/vcdbg log msg
001303.275: brfs: File read: /mfs/sd/config.txt
001303.735: brfs: File read: 229 bytes
001308.978: brfs: File read: /mfs/sd/dt-blob.bin
001337.037: HDMI0:EDID error reading EDID block 0 attempt 0
001338.305: HDMI0:EDID error reading EDID block 0 attempt 1
001339.569: HDMI0:EDID error reading EDID block 0 attempt 2
001340.835: HDMI0:EDID error reading EDID block 0 attempt 3
001342.100: HDMI0:EDID error reading EDID block 0 attempt 4
001343.366: HDMI0:EDID error reading EDID block 0 attempt 5
001344.630: HDMI0:EDID error reading EDID block 0 attempt 6
001345.896: HDMI0:EDID error reading EDID block 0 attempt 7
001347.161: HDMI0:EDID error reading EDID block 0 attempt 8
001348.427: HDMI0:EDID error reading EDID block 0 attempt 9
001349.449: HDMI0:EDID giving up on reading EDID block 0
001349.593: brfs: File read: 68470 bytes
001352.216: brfs: File read: /mfs/sd/config.txt
001352.724: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
001352.753: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
001365.656: gpioman: gpioman_get_pin_num: pin BT_ON not defined
001365.676: gpioman: gpioman_get_pin_num: pin WL_ON not defined
001392.400: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined

```

Am i misunderstanding what i should be expecting to see in the output? https://github.com/raspberrypi/firmware/issues/1118#issuecomment-1564291095 seems to imply that vclog is the replacement for vcdbg.

pelwell, 2023-08-04:
Strange - it's working for me in both 32-bit and 64-bit builds on a 32-bit and 64-bit OSs.

pelwell, 2023-08-04:
What does `od -Ax -tx1 /proc/device-tree/chosen/log` report?

vfazio, 2023-08-04:
> od -Ax -tx1 /proc/device-tree/chosen/lo

```
root@rpi-87fa00:~# od -Ax -tx1 /proc/device-tree/chosen/log
000000 3f f8 00 00 00 07 ff e0
000008

root@rpi-87fa00:~# busybox devmem 0x3ff80000
0x00000000

```

pelwell, 2023-08-05:
Is this running on a Pi 3 of some sort? vclog is lacking a trick that vcdbg has, namely the ability to use DMA to access the top 16MB of RAM that lives where the I/O space is mapped into the ARM address space. This space starts at 0x3f000000, and in this case the logging space starts at 0x3ff8000, which is well inside the 16MB blind spot of the ARM. Only Pi 3 (BCM2710) has this problem - older Pi 1, Pi 2, Zero and Zero 2 all have less RAM, and Pi 4 (BCM2711) puts the peripherals just below the 4GB boundary in the ARM address space, removing the blind spot.

Adding DMA support to vclog is a conceptually relatively small change, and fortunately the existing vclog implementation starts by making a copy of the buffer into some allocated ARM memory, so it shouldn't be to onerous a task.

vfazio, 2023-08-05:
Yes, I meant to add that yesterday, this is an RPi CM 3

pelwell, 2023-08-08:
> shouldn't be to onerous a task.

Ha - famous last words. It turned out not to be so straightforward, to the extent that it's been virtually reimplemented. I'm not quite ready to check in the changes, but I have a test build for you to try:

https://drive.google.com/file/d/1D3Bv3kL83-t3o_idDOEilxs5C9uHxFfK/view?usp=sharing

pelwell, 2023-08-08:
No problem. I forgot to say that it's a 32-bit static build so it _should_ run on any Pi.

pelwell, 2023-08-09:
FYI I've just uploaded a build of the release candidate - please use this version for your testing (the link is unchanged): https://drive.google.com/file/d/1D3Bv3kL83-t3o_idDOEilxs5C9uHxFfK/view?usp=sharing

uSpike, 2023-08-09:
Hi, I tried the RC on an RPi CM3+ and found that it doesn't seem to work:

```
root@rpi-da84b0:~# ./vclog -m
Could not map VC memory: Inappropriate ioctl for device
Fatal error: Cannot access logs
root@rpi-da84b0:~# ./vclog -a
Could not map VC memory: Inappropriate ioctl for device
Fatal error: Cannot access logs
```
```
root@rpi-da84b0:~# od -Ax -tx1 /proc/device-tree/chosen/log
000000 3f f8 00 00 00 07 ff e0
000008
root@rpi-da84b0:~# busybox devmem 0x3ff80000
0x00000000
```
please let me know if there's more info I can provide.

pelwell, 2023-08-09:
Unfortunately it relies on having a kernel DMA driver with an ioctl interface, which in this case is the old accelerated framebuffer driver. You won't get that with if the vc4 DRM drivers are enabled.

Does the old vcdbg work?

vfazio, 2023-08-09:
> Unfortunately it relies on having a kernel DMA driver with an ioctl interface, which in this case is the old accelerated framebuffer driver. You won't get that with if the vc4 DRM drivers are enabled.
> 
> Does the old vcdbg work?

It should. @uspike should be running a kernel and rootfs with modules built from the in tree bcmrpi3_defconfig from upstream rpi 5.15.y like I was. We build our own rootfs and have been using the static build but want to switch to open sources if possible. 

If there are additional modules necessary, can you call out the kconfig flags? This wasn't required for the old vcdbg method.

pelwell, 2023-08-10:
At least that makes sense - the difference is in the framebuffer driver. `bcm2708_fb` has the magic ioctl(), `simplefb` doesn't. This has always been the case, it's just that `bcm2708_fb` is being used less now we have working DRM drivers.

It may be that we have to graft the DMA support into another driver - vc_mem, perhaps - or find a way of disabling the accelerated framebuffer without sacrificing the DMA support.

vfazio, 2023-08-10:
Well, it's probably pretty apparent how often we use this utility if it's just now coming up. Maybe I don't worry about it for now as we're pending a migration to 6.1 at some point but that has a rather large testing surface area so it won't be soon. 

I do have FB_BCM2708 built into this kernel (it's in the default config). is there other magic necessary?

```
root@rpi-87fa00:~# zcat /proc/config.gz | grep BCM2708
CONFIG_BCM2708_VCMEM=y
CONFIG_I2C_BCM2708=m
CONFIG_I2C_BCM2708_BAUDRATE=100000
CONFIG_FB_BCM2708=y

```

vfazio, 2023-08-10:
Also, i think it's fair to say that your changes _work_, which was the original scope of this issue. I don't want to derail this too much but i think the additional conversation may be useful for other users.

pelwell, 2023-08-10:
The module is included in all our kernel builds, but using either `dtoverlay=vc4-kms-v3d` or `dtoverlay=vc4-fkms-v3d` prevents the device from being instantiated. The firmware detects that, and enables the `simplefb` driver in its place. I suspect we would end up with:
1. an extension to the bcm2708_fb driver that just presents the DMA interface (perhaps triggered by a different compatible string),
2. modified overlays that create an instance of the new DMA-only framebuffer as well as disabling the old one, and
3. a slight tweak to vclog that looks for a different /dev/ name.

Given that list, extending vc_mem sounds easier.

pelwell, 2023-08-10:
By the way, if you aren't particularly memory constrained in your application you can bypass all this nastiness (at least while your debugging) by putting `total_mem=1008` in `config.txt`...

vfazio, 2023-08-10:
Also, for posterity, our config.txt. we do not enable those overlays:

```
root@rpi-87fa00:/mnt/persistent# cat /boot/config.txt
dtparam=audio=on
dtparam=spi=on
dtparam=i2c_arm=on
dtparam=i2c1=on
dtoverlay=disable-bt
dtoverlay=disable-wifi
kernel=u-boot.bin
enable_uart=1
arm_64bit=1
start_x=1

[pi3+]
# Model 3B
[board-type=0x8]
cmdline=cmdline-noserial.txt

```

pelwell, 2023-08-10:
> this is just chopping off an additional 16MB of RAM in our case?
Yes it is.

vfazio, 2023-08-10:
> The module is included in all our kernel builds, but using either `dtoverlay=vc4-kms-v3d` or `dtoverlay=vc4-fkms-v3d` prevents the device from being instantiated. The firmware detects that, and [...]

@pelwell are there additional tasks you want to drive from this issue? Or has everything been implemented to your satisfaction in [94b89a6](https://github.com/raspberrypi/utils/commit/94b89a69cdf3d10cbe8ee7571584950e5df768f5) ?

Also, as i mentioned, we do not load the overlays you mentioned, but i assume we're still falling victim to the `simplefb` driver being loaded and not being able to service this ioctl.

pelwell, 2023-08-10:
I think we can close the issue, but I've made a note of the vcdbg/vclog vs kms driver issue.

I haven't figured out why you don't have the frame buffer instantiated - ooh, dos It

closed 2023-08-10 by pelwell

vfazio, 2023-08-21:
> I think we can close the issue, but I've made a note of the vcdbg/vclog vs kms driver issue.
> 
> I haven't figured out why you don't have the frame buffer instantiated, but I'm suspicious about u-boot's role in dtb handling.

I haven't chased it down yet, but i can confirm that if i modify my config.txt:

```
root@rpi-87fa00:~# uname -a
Linux rpi-87fa00 5.15.84-v8 #1 SMP PREEMPT Fri Aug 18 20:37:52 UTC 2023 aarch64 GNU/Linux
root@rpi-87fa00:~# cat /boot/config.txt
dtparam=audio=on
dtparam=spi=on
dtparam=i2c_arm=on
dtparam=i2c1=on
dtoverlay=disable-bt
dtoverlay=disable-wifi
kernel=Image
enable_uart=1
arm_64bit=1
start_x=1
cmdline=cmdline.txt

[pi3+]
# Model 3B
[board-type=0x8]
cmdline=cmdline-noserial.txt
root@rpi-87fa00:~# cat /boot/cmdline.txt
console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 rootfstype=ext4 rootwait root=/dev/mmcblk0p5 cma=64M

```

That things seem to work...

```
root@rpi-87fa00:~# vclog -m | head -n20
001262.506: brfs: File read: /mfs/sd/config.txt
001262.986: brfs: File read: 245 bytes
001268.448: brfs: File read: /mfs/sd/dt-blob.bin
001296.651: HDMI0:EDID error reading EDID block 0 attempt 0
001297.919: HDMI0:EDID error reading EDID block 0 attempt 1
001299.183: HDMI0:EDID error reading EDID block 0 attempt 2
001300.449: HDMI0:EDID error reading EDID block 0 attempt 3
001301.714: HDMI0:EDID error reading EDID block 0 attempt 4
001302.980: HDMI0:EDID error reading EDID block 0 attempt 5
001304.244: HDMI0:EDID error reading EDID block 0 attempt 6
001305.510: HDMI0:EDID error reading EDID block 0 attempt 7
001306.775: HDMI0:EDID error reading EDID block 0 attempt 8
001308.041: HDMI0:EDID error reading EDID block 0 attempt 9
001309.063: HDMI0:EDID giving up on reading EDID block 0
001309.211: brfs: File read: 68470 bytes
001311.975: brfs: File read: /mfs/sd/config.txt
001312.504: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
001312.534: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
001325.337: gpioman: gpioman_get_pin_num: pin BT_ON not defined
001325.357: gpioman: gpioman_get_pin_num: pin WL_ON not defined

```

So there's something about the handoff from firmare to uboot to linux causing the problem.

I dumped /proc/device-tree between the two:

```diff
root@rpi-87fa00:/mnt/persistent# diff linux_dev_tree.dts uboot_dev_tree.dts
1116c1116,1117
<               bootargs = "coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:87:FA:00 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 rootfstype=ext4 rootwait root=/dev/mmcblk0p5 cma=64M";
---
>               u-boot,version = "2023.01";
>               bootargs = "coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:87:FA:00 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 rootfstype=ext4 rootwait root=/dev/mmcblk0p5 cma=64M root=/dev/mmcblk0p5 cma=64M";
1134a1136,1145
>       };
>
>       framebuffer@3eadf000 {
>               format = "a8r8g8b8";
>               height = <0x1a0>;
>               width = <0x290>;
>               stride = <0xa40>;
>               compatible = "simple-framebuffer";
>               status = "okay";
>               reg = <0x3eadf000 0x10a800>;

```

I'm guessing it's that framebuffer node causing the grief?

vfazio, 2023-08-21:
For posterity, the framebuffer node gets added here: https://source.denx.de/u-boot/u-boot/-/blob/master/board/raspberrypi/rpi/rpi.c#L563

This "fixup" can be avoided if `CONFIG_OF_BOARD_SETUP` is not set.