raspberrypi/utils #33: pinctrl: Failed to mmap gpiolib

macmpi, 2023-09-14:
On PiZeroW, kernel 6.1.27-0-rpi, musl (Alpine)
```
# pinctrl
Failed to mmap gpiolib - Operation not permitted

# pinctrl get
Failed to mmap gpiolib - Operation not permitted

# pinctrl funcs 9-11
9, SPI_MISO/GPIO9, SPI0_MISO, SD1, DPI_D5, AVEOUT_VID5, AVEIN_VID5, -
10, SPI_MOSI/GPIO10, SPI0_MOSI, SD2, DPI_D6, AVEOUT_VID6, AVEIN_VID6, -
11, SPI_SCLK/GPIO11, SPI0_SCLK, SD3, DPI_D7, AVEOUT_VID7, AVEIN_VID7, -

```

pelwell, 2023-09-15:
Cool - the first pinctrl issue.
1. Can I assume you're running as root?
2. What does `groups` report?
3. What does `ls -l /dev/mem` report?

macmpi, 2023-09-15:
:wink: 
Yes running as root for now.
```
# groups
root bin daemon sys adm disk wheel floppy dialout tape video
# ls -l /dev/mem
crw-r-----    1 root     root        1,   1 Jan  1  1970 /dev/mem

```

Maybe empty arguments shortcut commands ([funcs](https://github.com/raspberrypi/utils/blob/master/pinctrl/pinctrl.c#L588)) do not follow proper flow?

Is `pinctrl` scheduled to replace `raspi-gpio` in `raspinfo`?

pelwell, 2023-09-15:
It's working for me on a Zero W, so there must be something different about your environment - probably the kernel. Can you upload /proc/config.gz somewhere (you may need to "modprobe configs" first).

The `funcs` option does not require root privilege because it's essentially just documentation about the chip.

> Is pinctrl scheduled to replace raspi-gpio in raspinfo?

It's not scheduled.

macmpi, 2023-09-15:
here you go: [config-rpi.txt](https://github.com/raspberrypi/utils/files/12618497/config-rpi.txt)

> It's not scheduled.

Would be neat though, to have everything in same spot.
Or maybe as an alternative, as with `vcdbg`/`vclog`

pelwell, 2023-09-15:
Your config includes `CONFIG_EXCLUSIVE_SYSTEM_RAM=y` - can you try with that removed?

pelwell, 2023-09-15:
Scratch that - try without `CONFIG_IO_STRICT_DEVMEM=y` instead.

macmpi, 2023-09-15:
hum...I'm afraid I'm not set-up to recompile the kernel at that time...
and I'm not sure I'll be able to advocate such change to be integrated in Alpine distro overall: that's the only option?

pelwell, 2023-09-15:
Yes, at least until such time as pinctrl is made to support /dev/gpiomem.

pelwell, 2023-09-15:
I think there _is_ another way - put `iomem=relaxed` in cmdline.txt.

macmpi, 2023-09-15:
:+1: 
Thanks. May be worth mentioning it in readme then.
```
# pinctrl
 0: ip    -- | hi // ID_SDA/GPIO0 = input
 1: ip    -- | hi // ID_SCL/GPIO1 = input
 2: ip    -- | hi // SDA1/GPIO2 = input
 3: ip    -- | hi // SCL1/GPIO3 = input
 4: ip    -- | hi // GPIO_GCLK/GPIO4 = input
 5: ip    -- | hi // GPIO5 = input
 6: ip    -- | hi // GPIO6 = input
 7: ip    -- | hi // SPI_CE1_N/GPIO7 = input
 8: ip    -- | hi // SPI_CE0_N/GPIO8 = input
 9: ip    -- | lo // SPI_MISO/GPIO9 = input
10: ip    -- | lo // SPI_MOSI/GPIO10 = input
11: ip    -- | lo // SPI_SCLK/GPIO11 = input
12: ip    -- | lo // GPIO12 = input
13: ip    -- | lo // GPIO13 = input
14: ip    -- | lo // TXD1/GPIO14 = input
15: ip    -- | hi // RXD1/GPIO15 = input
16: ip    -- | lo // GPIO16 = input
17: ip    -- | lo // GPIO17 = input
18: ip    -- | lo // GPIO18 = input
19: ip    -- | lo // GPIO19 = input
20: ip    -- | lo // GPIO20 = input
21: ip    -- | lo // GPIO21 = input
22: ip    -- | lo // GPIO22 = input
23: ip    -- | lo // GPIO23 = input
24: ip    -- | lo // GPIO24 = input
25: ip    -- | lo // GPIO25 = input
26: ip    -- | lo // GPIO26 = input
27: ip    -- | lo // GPIO27 = input
28: ip    -- | hi // SDA0/GPIO28 = input
29: ip    -- | hi // SCL0/GPIO29 = input
30: a3    -- | lo // CTS0/GPIO30 = CTS0
31: a3    -- | lo // RTS0/GPIO31 = RTS0
32: a3    -- | hi // TXD0/GPIO32 = TXD0
33: a3    -- | hi // RXD0/GPIO33 = RXD0
34: a3    -- | lo // SD1_CLK/GPIO34 = SD1_CLK
35: a3    -- | hi // SD1_CMD/GPIO35 = SD1_CMD
36: a3    -- | hi // SD1_DATA0/GPIO36 = SD1_DAT0
37: a3    -- | hi // SD1_DATA1/GPIO37 = SD1_DAT1
38: a3    -- | hi // SD1_DATA2/GPIO38 = SD1_DAT2
39: a3    -- | hi // SD1_DATA3/GPIO39 = SD1_DAT3
40: op -- -- | lo // CAM_GPIO1/GPIO40 = output
41: op -- -- | hi // WL_ON/GPIO41 = output
42: ip    -- | lo // NC/GPIO42 = input
43: a0    -- | lo // WIFI_CLK/GPIO43 = GPCLK2
44: op -- -- | lo // CAM_GPIO0/GPIO44 = output
45: op -- -- | hi // BT_ON/GPIO45 = output
46: ip    -- | hi // HDMI_HPD_N/GPIO46 = input
47: op -- -- | lo // STATUS_LED_N/GPIO47 = output
48: a0    -- | lo // SD_CLK_R/GPIO48 = SD0_CLK
49: a0    -- | hi // SD_CMD_R/GPIO49 = SD0_CMD
50: a0    -- | hi // SD_DATA0_R/GPIO50 = SD0_DAT0
51: a0    -- | hi // SD_DATA1_R/GPIO51 = SD0_DAT1
52: a0    -- | hi // SD_DATA2_R/GPIO52 = SD0_DAT2
53: a0    -- | hi // SD_DATA3_R/GPIO53 = SD0_DAT3
```

pelwell, 2023-09-15:
> at least until such time as pinctrl is made to support /dev/gpiomem.

That time came sooner than expected - grab the latest patch and rebuild. If your system has /dev/gpiomem then you won't need the `iomem=relaxed` option. If /dev/gpiomem has suitable permissions then you won't even need root privilege.

macmpi, 2023-09-15:
awesome, that was quick: works fine!
Thanks!

```
# ls -la /dev/gpiomem
crw-rw----    1 root     root      245,   0 Jan  1  1970 /dev/gpiomem
```

pelwell, 2023-09-15:
FYI, in RPiOS we have:
```
$ ls -l /dev/gpiomem
crw-rw---- 1 root gpio 245, 0 Sep 15 17:39 /dev/gpiomem
$ groups 
pi adm dialout cdrom sudo audio video plugdev games users input render netdev lpadmin gpio i2c spi
```

macmpi, 2023-09-16:
Thanks. Indeed maybe adding `gpio` group and related rule in Alpine could be beneficial too.

closed 2023-09-16 by macmpi