raspberrypi/utils #52: RaspberryPi 4 - ArchlinuxARM: Can't open device file: /dev/vcio

danieletorelli, 2023-10-26:
Hello, I'm experiencing this issue on my RaspberryPi 4 since Archlinux ARM started using those binaries some day ago as `/dev/vcio` doesn't seem to be created anymore.

**Kernel:**
```
Linux alarm 6.2.10-1-aarch64-ARCH #1 SMP PREEMPT_DYNAMIC Fri Apr  7 10:32:52 MDT 2023 aarch64 GNU/Linux
```

**Steps to reproduce:**
```
~ # vcgencmd 
Can't open device file: /dev/vcio
Try creating a device file with: sudo mknod /dev/vcio c 100 0
```

graysky2, 2023-10-26:
I suspect your issue is that you're running the vanilla linux kernel (linux-aarch64) and not the RPi Foundation based kernel (linux-rpi).

danieletorelli, 2023-10-27:
Your suspect is right.

Before it was all working well with the vanilla kernel thanks probably to `raspberrypi-firmware` package, that has been deprecated 3 days ago in favor of `raspberrypi-utils` from this repo.

Switching for `linux-rpi` kernel fixed this issue.

closed 2023-10-27 by danieletorelli

graysky2, 2023-10-27:
Honestly don't know about vanilla kernel.  If you want cool stuff on the RPi platform like hw decoding, use linux-rpi.

pathcheatsheet, 2023-11-08:
Hello,

Looks like I am having a very similar issue.
I am trying to get a set of python scripts (pipresents) running on a RPi 4B.
I do not know Python but the code seems to execute vcgencmd and throws an error. When I run vcgencmd as non-root, it throws OP's error.  (Tried this on Arch ARM aarch64 as well as armv7h; both running linux-rpi; however it does work on RPi OS but I don't want to use a Debian-based system.)
It looks like there is a permission problem as it does what it has to as root:
```
[szg@alarm ~]$ uname -a
Linux alarm 6.1.61-2-rpi-ARCH #1 SMP PREEMPT Mon Nov  6 08:36:16 MST 2023 aarch64 GNU/Linux
[szg@alarm ~]$ vcgencmd get_mem gpu
Can't open device file: /dev/vcio
Try creating a device file with: sudo mknod /dev/vcio c 100 0
[szg@alarm ~]$ sudo vcgencmd get_mem gpu
gpu=76M
```
Exactly the same on armv7h. 
Not sure whether this is expected behaviour since vcgencmd does work under normal user on RPi OS Bullseye:
```
szg@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
szg@raspberrypi:~ $ vcgencmd get_mem gpu
gpu=76M
```

pathcheatsheet, 2023-11-08:
A quick update: booted back into Arch armv7 and added the szg user to the video group there. This does not help, even though permissions for /dev/vcio are identical in both OSes:
```
crw-rw---- 1 root video 10, 125 Nov  6 18:27 /dev/vcio
```

danieletorelli, 2023-11-09:
@pathcheatsheet In my case it was enough to add my user to the `video` group, disconnect and reconnect from ssh. I was able to run this after reconnecting:

```
alarm ~ $ vcgencmd get_mem gpu
gpu=256M

loki ~ $ id
uid=1000(alarm) gid=984(users) groups=984(users),985(video),998(wheel)
```