raspberrypi/utils #62: eeptools build errors

macmpi, 2024-01-12:
```
ninja: job failed: /usr/bin/gcc   -Os -fstack-clash-protection -Wformat -Werror=format-security -Wall -Wextra -Werror -Os -DNDEBUG   -ffunction-sections -MD -MT eeptools/CMakeFiles/eepmake.dir/eeplib.c.o -MF eeptools/CMakeFiles/eepmake.dir/eeplib.c.o.d -o eeptools/CMakeFiles/eepmake.dir/eeplib.c.o -c /eeptools/eeplib.c
In file included from /eeptools/eeplib.c:5:
/eeptools/eeplib.c: In function 'eepio_start':
/eeptools/eeplib.h:26:21: error: implicit declaration of function 'be32toh' [-Werror=implicit-function-declaration]
   26 | #define HEADER_SIGN be32toh((((char)'R' << 24) | ((char)'-' << 16) | ((char)'P' << 8) | ((char)'i')))
      |                     ^~~~~~~
/eeptools/eeplib.c:194:40: note: in expansion of macro 'HEADER_SIGN'
  194 |                 eep_header.signature = HEADER_SIGN;
      |                                        ^~~~~~~~~~~
/eeptools/eeplib.c:206:42: error: comparison of integer expressions of different signedness: 'uint32_t' {aka 'unsigned int'} and 'int' [-Werror=sign-compare]
  206 |                 if (eep_header.signature != HEADER_SIGN)
      |                                          ^~
/eeptools/eeplib.c: In function 'eepio_end':
/eeptools/eeplib.c:237:25: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  237 |                 if (pos != eep_header.eeplen)
      |                         ^~
/eeptools/eeplib.c:239:29: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  239 |                 if (pos_end != eep_header.eeplen)
      |                             ^~
/eeptools/eeplib.c:241:29: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  241 |                 if (pos_end != eep_header.eeplen)
      |                             ^~
/eeptools/eeplib.c: In function 'eepio_atom_end':
/eeptools/eeplib.c:287:49: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  287 |                 if (pos - eepio_atom_data_start != eep_atom_header.dlen)
      |                                                 ^~
cc1: all warnings being treated as errors
ninja: job failed: /usr/bin/gcc   -Os -fstack-clash-protection -Wformat -Werror=format-security -Wall -Wextra -Werror -Os -DNDEBUG   -ffunction-sections -MD -MT eeptools/CMakeFiles/eepdump.dir/eeplib.c.o -MF eeptools/CMakeFiles/eepdump.dir/eeplib.c.o.d -o eeptools/CMakeFiles/eepdump.dir/eeplib.c.o -c /eeptools/eeplib.c
In file included from /eeptools/eeplib.c:5:
/eeptools/eeplib.c: In function 'eepio_start':
/eeptools/eeplib.h:26:21: error: implicit declaration of function 'be32toh' [-Werror=implicit-function-declaration]
   26 | #define HEADER_SIGN be32toh((((char)'R' << 24) | ((char)'-' << 16) | ((char)'P' << 8) | ((char)'i')))
      |                     ^~~~~~~
/eeptools/eeplib.c:194:40: note: in expansion of macro 'HEADER_SIGN'
  194 |                 eep_header.signature = HEADER_SIGN;
      |                                        ^~~~~~~~~~~
/eeptools/eeplib.c:206:42: error: comparison of integer expressions of different signedness: 'uint32_t' {aka 'unsigned int'} and 'int' [-Werror=sign-compare]
  206 |                 if (eep_header.signature != HEADER_SIGN)
      |                                          ^~
/eeptools/eeplib.c: In function 'eepio_end':
/eeptools/eeplib.c:237:25: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  237 |                 if (pos != eep_header.eeplen)
      |                         ^~
/eeptools/eeplib.c:239:29: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  239 |                 if (pos_end != eep_header.eeplen)
      |                             ^~
/eeptools/eeplib.c:241:29: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  241 |                 if (pos_end != eep_header.eeplen)
      |                             ^~
/eeptools/eeplib.c: In function 'eepio_atom_end':
/eeptools/eeplib.c:287:49: error: comparison of integer expressions of different signedness: 'long int' and 'uint32_t' {aka 'unsigned int'} [-Werror=sign-compare]
  287 |                 if (pos - eepio_atom_data_start != eep_atom_header.dlen)
      |                                                 ^~
cc1: all warnings being treated as errors
ninja: subcommands failed
```

pelwell, 2024-01-15:
Which version of GCC is this? gcc-12 isn't complaining for me.
How are you building it using Ninja?

macmpi, 2024-01-15:
it is built by Alpine [CI tools](https://gitlab.alpinelinux.org/macmpi/aports/-/blob/1e4d0816573d0a73d30a60b3d1a7490115abc3e7/community/raspberrypi-utils/APKBUILD): log is [here](https://gitlab.alpinelinux.org/macmpi/aports/-/jobs/1241792).
seemingly GCC 13.2.1

macmpi, 2024-01-15:
looks fine yes :+1: 
Thanks.

closed 2024-01-15 by macmpi