raspberrypi/utils #66: thread concurrency for gpio_get_level()

rafael2k, 2024-01-17:
Considering concurrency, I'm already wrapping all calls that change gpio states with mutex.

But considering gpio_get_level(), could it be called concurrently in a multi-thread software or better I just make sure I don't call API concurrently?

Reading the datasheet it seems to me that DMA memory writes to gpio registers on memory are atomic.

pelwell, 2024-01-17:
> Reading the datasheet it seems to me that DMA memory writes to gpio registers on memory are atomic.

There's no DMA involved here, but apart from the possibility that another thread has changed the direction or FSEL of your pin there should be no atomicity problems for `gpio_get_level` and `gpio_set_drive`  without an external mutex.

rafael2k, 2024-01-17:
Btw, how distributions should package Raspberry Pi's gpiolib? Do you think a system-wide shared object is a good idea? 

GPIOLIB is just great, perfect for people moving away from WiringPi (like me).
: )

pelwell, 2024-01-18:
> Btw, how distributions should package Raspberry Pi's gpiolib?

It wasn't really intended to be a WiringPi replacement - we'd prefer that to be libgpiod - but I thought it may have other uses. If it is going to be used by more than one binary then a shared library seems sensible, but static builds are so much easier to maintain.

rafael2k, 2024-01-19:
Tbh, I found your library easier to use than libgpiod, especially when the target hardware is Rasperry Pi computer line.
: )
Anyway, when I create a "debian/" folder for package creation I let you know, in case you have any concern. Thanks a lot!

I'll close this one.

closed 2024-01-23 by rafael2k