raspberrypi/utils #73: GPIOLIB much slower than WiringPi on Pi 5

rafael2k, 2024-03-12:
This is not a bug, but I'd like some light to understand why the very recent Pi 5 support in WiringPi for GPIO access is so much faster than GPIOLIB one. For eg., setting drive low and high in WiringPi reaches 20 MHz, while with GPIOLIB, not even 0.5 MHz.

Sample code of what I'm running to compare here:
https://github.com/rafael2k/Benchmark

rafael2k, 2024-03-13:
Wow, what a difference!! Thanks @pelwell!
Now I get:
   240000000 toggle took 15.519 s, Time per toggle 0.065 us, Freq 15.465 MHz 
basically, 30x faster.
: )

pelwell, 2024-03-13:
Cool - I may tidy it up a bit, but I'll make sure that logic becomes standard.

pelwell, 2024-03-13:
#74 is a slightly neater implementation. Would you mind testing it as well?

rafael2k, 2024-03-13:
Sure. Will come back soon with results.

rafael2k, 2024-03-13:
It keeps working, with approximately the same speed of the patch you posted.
 240000000 toggle took 15.267 s, Time per toggle 0.064 us, Freq 15.721 MHz

pelwell, 2024-03-13:
Thanks - that's merged now.

rafael2k, 2024-03-13:
Cheers!

closed 2024-03-13 by pelwell