raspberrypi/utils #76: Is it possible to read values from ADCs with pinctrl?

rafael2k, 2024-04-07:
I realize that with vcgencmd I can read some ADC values in the Pi 5. Can the same be done through pinctrl?

pelwell, 2024-04-08:
Rafael, the short answer is no. The slightly longer answer is no and, even if you could, it wouldn't be of much use unless you want to monitor VBUS because they are dedicated pins on RP1 (you can't enable ADC on regular GPIOs) and the Pi 5 board design doesn't make the spare channels available.

If you are still curious, the ADC values can be read from sysfs:
```
$ grep . /sys/class/hwmon/hwmon1/in*_input
```

rafael2k, 2024-04-08:
Thanks @pelwell! Among these 4 ADC inpu6ts, are any of those available to the outside world (even if it involves reworking in the board - for experimentation purposes)?

pelwell, 2024-04-08:
Define available.

rafael2k, 2024-04-08:
Some pin in RP1 I can tap / solder to outside the chip.

pelwell, 2024-04-08:
The unused channels will be totally unconnected - just balls on the bottom of the chip. The used channels have pull-up and pull-down resistors that would have to be located and removed - I have schematics, but not the layout.

pelwell, 2024-04-08:
No they aren't.

pelwell, 2024-04-08:
I believe that is the case. The bottom of the board under RP1 is so congested that it would be counterproductive to connect any "pins" that aren't needed.

rafael2k, 2024-04-08:
Where can I find the list of the the used ADC channels? Are they all listed in "vcgencmd pmic_read_adc" output?

pelwell, 2024-04-08:
The PMIC ADC channels are completely separate - they provide feedback on the various power rails. As I said earlier, the RP1 ADCs are readable from `/sys/class/hwmon/hwmon1/in*_input`.

rafael2k, 2024-04-08:
Sorry the mix-up from my side. So in your opinion, is having an externally accessible ADC in the Pi 5, be it using any PMIC ADC or RP1 ADC, something feasible to pursue using amateur-grade lab equipment? 

Or better I just keep using an ATTiny for my analog readings, connected over i2c...

pelwell, 2024-04-08:
Although much is possible with enough information and soldering skills, I can't recommend butchering a Pi 5 in a way I wouldn't feel comfortable doing myself (or asking one of our hardware guys to do for me).

rafael2k, 2024-04-08:
Fair enough. Thanks @pelwell!
: )

rafael2k, 2024-04-08:
Last question - Which is the purpose of the used RP1 ADC lines? 

These are the ones I can read from these device files, right?
/sys/class/hwmon/hwmon1/in*_input

pelwell, 2024-04-08:
VBUS monitoring, and RAM size detection.

closed 2024-04-08 by pelwell