raspberrypi/utils #78: pinctrl help - no way to set ALT0 function (doc or indexing issue)

drankinatty, 2024-04-10:
`pinctrl help` lists the valid options for `set` as:

```none
...
Valid [options] for pinctrl set are:
  ip      set GPIO as input
  op      set GPIO as output
  a1-a7   set GPIO to fsel in the range 1-7
  no      set GPIO to no function (NONE)
  pu      set GPIO in-pad pull up
  pd      set GPIO pin-pad pull down
  pn      set GPIO pull none (no pull)
  dh      set GPIO to drive high (1) level (only valid if set to be an output)
  dl      set GPIO to drive low (0) level (only valid if set to be an output)
...
```

However, the valid `ALTx` functions on the Pi are indexed `0-5`. I have not been through the code to determine if this is simply a documentation issue or an indexing issue, but on first impression there is no way to set the `a0` function for any pin. Displaying current `get` information discloses the `a0` functions in use on many pins, e.g.

```none
$ pinctrl get 7-11
 7: op -- -- | hi // SPI_CE1_N/GPIO7 = output
 8: op -- -- | hi // SPI_CE0_N/GPIO8 = output
 9: a0    -- | lo // SPI_MISO/GPIO9 = SPI0_MISO
10: a0    -- | lo // SPI_MOSI/GPIO10 = SPI0_MOSI
11: a0    -- | lo // SPI_SCLK/GPIO11 = SPI0_SCLK
```

If there is no valid `a0` option for `pinctrl`, just how does one go about restoring an `a0` function?

Conversely, `raspi-gpio help` lists the `set` options with the `0-5` ALT functions available, e.g.

```none
...
Valid [options] for raspi-gpio set are:
  ip      set GPIO as input
  op      set GPIO as output
  a0-a5   set GPIO to alternate function alt0-alt5
  pu      set GPIO in-pad pull up
  pd      set GPIO pin-pad pull down
  pn      set GPIO pull none (no pull)
  dh      set GPIO to drive to high (1) level (only valid if set to be an output)
  dl      set GPIO to drive low (0) level (only valid if set to be an output)
...
```

Seems a quick check on what `pinctrl` does for `a0` is in order. This is a great little set of utilities!

pelwell, 2024-04-10:
The help text is wrong - values `a0` to `a8` are recognised, but not valid on all platforms.

pelwell, 2024-04-14:
Fixed by https://github.com/raspberrypi/utils/commit/8fafd49da0d91665778cb698f49c83f5be793d86.

closed 2024-04-14 by pelwell

drankinatty, 2024-04-18:
Thank you. `pinctrl` is a great little tool!