raspberrypi/utils #82: Add option to show only 0 or 1 for level

TheRaven500, 2024-05-11:
Is it possible to add option to only show 0 or 1 for the status?
I work very often with bash scripts and GPIOs. In this scripts i need the level of the GPIO input (0 or 1).
ATM i do it like this:
`pinctrl get $SWITCHPIN|awk '{print $3}'|grep -o '.\{1\}$'`
It works, but it is ugly. :smile:
Maybe something like:
`pinctrl get $SWITCHPIN level`
Or "lo" for LevelOnly:
`pinctrl get $SWITCHPIN lo`
But this sounds also a bit like "low".

pelwell, 2024-05-11:
That's easy enough, but I'd make it a switch, e.g. `-l`:
```
pinctrl -l get $SWITCHPIN
```

pelwell, 2024-05-11:
Slight change of plan - it's now `pinctrl level $SWITCHPIN` (or `pinctrl lev $SWITCHPIN`). Levels for multiple GPIOs are returned separated by spaces.

See: https://github.com/raspberrypi/utils/pull/83

TheRaven500, 2024-05-11:
Ok, you are very fast! :smile: I will try it as soon as possible. But i am sure it will work perfect. Thanks alot for your work.

TheRaven500, 2024-05-14:
How can i test it? 🤔

pelwell, 2024-05-14:
Sorry - GitHub doesn't send notifications for comment edits.

You're going to have to build it yourself, but it is quick and easy (honestly):
1. Click on the link to the pull request in my comment.
2. At the top of that PR where it says "pelwell wants to merge 1 commit into raspberrypi:master from pelwell:is82", click on "pelwell:is82".
3. On the next page, click on the green Code button, make sure the HTTPS tab is open, and copy the URL (or click the icon on the right to put it in the clipboard).
4. In a terminal on your Pi 5 (if you aren't there already), navigate to somewhere where you are happy to create a `utils` directory, and run the following:
```
git clone -b is82 https://github.com/pelwell/utils.git
cd utils
sudo apt install cmake
cmake .
make
sudo make install
```
And that should be it.

TheRaven500, 2024-05-14:
Because of that i made a "double post" sorry. :wink:
Thank you very much, i had to add "libfdt-dev" also for building:
```
git clone -b is82 https://github.com/pelwell/utils.git
cd utils
sudo apt install cmake libfdt-dev
cmake .
make
sudo make install
```
Works and looks good! :sunglasses: 
Two things:
1. Maybe adding an "Example" of level in the "help" would be nice. It makes it more clear how to use it.
2. On the beginning of level there is a " " blank. It gives " 1" back and not "1". Is that intentional?

pelwell, 2024-05-14:
1. I could add an example, but that help is already getting quite long. Perhaps.
2. It's expected rather than intentional. To get the ideal output for two gpios of `0 1`, something has to treat the first (or last) element specially. I opted for `0 1` to keep it simple , but it could also be `0 1 `.  Hmmph - I'll give it some thought.

TheRaven500, 2024-05-14:
I think in most cases you need only one output. As explained in the other discussion, i am searching an alternative to replace sysfs in bash scripts. I do just something like:
`if [ "$(pinctrl lev ${GPIOPIN})" = "1" ]; then`
So for me it is better to have just "1" and not " 1". But you are the expert and know what is the better option. :smiley: 
I can also live with " 1".

pelwell, 2024-05-14:
You've acted as the voice of my conscience, and I've now done it right. To download and build the changes, starting in the utils directory:
```
git fetch origin
git reset --hard origin/is82
make
sudo make install
```

TheRaven500, 2024-05-14:
What has changed? :thinking:

pelwell, 2024-05-14:
The spaces should only appear between entries - if you aren't seeing this then your update hasn't worked.

TheRaven500, 2024-05-14:
Looks like something has not worked. :wink: It is still " 1".

pelwell, 2024-05-14:
OK - it's going to be hard to figure out what has failed, so we can just delete it all and start again. Running from the utils directory:
```
cd ..
rm -rf utils
git clone -b is82 https://github.com/pelwell/utils.git
cd utils
cmake .
make
sudo make install
```

TheRaven500, 2024-05-14:
Does it work for you? Because for me not. Still " 1". Strange... :thinking: 
I also tried it to run it directly from the build path:
`/tmp/utils/pinctrl/pinctrl lev 12`

TheRaven500, 2024-05-14:
Now it looks ok (or maybe perfect) :+1: :smiley:

pelwell, 2024-05-14:
That's merged now.

TheRaven500, 2024-05-14:
Ok, i think then it is fixed. Thank you very much for your time and your patience. :+1:

closed 2024-05-14 by TheRaven500