raspberrypi/utils #95: the pull request I submitted?

seamusdemora, 2024-08-20:
Hello - Yesterday, I submitted a very simple pull request against `pinctrl`. As far as I could see in GitHub, it was sent OK.

But it's not showing up here under the pull requests... can someone tell me what happened?

pelwell, 2024-08-20:
You must have imagined it - there's no evidence for its existence: https://github.com/raspberrypi/utils/issues?q=
(Note the lack of gaps in the numbering)

macmpi, 2024-08-20:
that one? https://github.com/raspberrypi/utils/pull/59

seamusdemora, 2024-08-20:
@macmpi :  

Yes - but it's odd. The pull I asked about here (for pinctrl) (#95) was combined with (#59) (for raspinfo). I didn't notice that yesterday. Seems odd it would combine them. 

Anyway - I guess you've got both of them now - so you can throw them both out at one time :)