raspberrypi/utils #101: Compile error on OpenWrt build system for RPi5B (GCC 14.2.0)

graysky2, 2024-10-29:
Trying to build the latest commit results in the following error:
```
make[1]: Entering directory '/scratch/union'
make[2]: Entering directory '/scratch/union/package/utils/bcm27xx-utils'
touch /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.prepared_check
. /scratch/union/include/shell.sh; zstdcat /scratch/union/dl/bcm27xx-utils-2024.04.24.tar.zst | tar -C /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24
rm -rf /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/patches; mkdir -p /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/patches
touch /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.quilt_used
touch /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.prepared
if [ -s "/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/patches/series" ]; then (cd "/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24"; if quilt --quiltrc=- next >/dev/null 2>&1; then quilt --quiltrc=- push -a; else quilt --quiltrc=- top >/dev/null 2>&1; fi ); fi
Applying patch 0001-raspinfo-adapt-to-OpenWrt.patch
patching file raspinfo/raspinfo

Now at patch 0001-raspinfo-adapt-to-OpenWrt.patch
touch "/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.quilt_checked"
rm -f /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.configured_*
rm -f /scratch/union/staging_dir/target-aarch64_cortex-a76_musl/stamp/.bcm27xx-utils_installed
mkdir -p /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24
(cd /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24; CFLAGS="-mcpu=cortex-a76 -O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24=bcm27xx-utils-2024.04.24 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-mcpu=cortex-a76 -O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24=bcm27xx-utils-2024.04.24 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/usr/lib -L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/lib -fuse-ld=bfd -znow -zrelro " cmake --no-warn-unused-cli -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=aarch64 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER_LAUNCHER="" -DCMAKE_C_COMPILER="/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_CXX_COMPILER_LAUNCHER="" -DCMAKE_CXX_COMPILER="/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-g++" -DCMAKE_ASM_COMPILER_LAUNCHER="" -DCMAKE_ASM_COMPILER="/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/usr/lib -L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/lib -fuse-ld=bfd -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/usr/lib -L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/lib -fuse-ld=bfd -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/usr/lib -L/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/lib -fuse-ld=bfd -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/scratch/union/staging_dir/target-aarch64_cortex-a76_musl/usr;/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/scratch/union/staging_dir/target-aarch64_cortex-a76_musl -DCMAKE_PREFIX_PATH=/scratch/union/staging_dir/target-aarch64_cortex-a76_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE -DCMAKE_GENERATOR="Ninja" /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24 )
Not searching for unused variables given on the command line.
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24
touch /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.built
touch /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.built_check
MAKEFLAGS="--jobserver-auth=fifo:/tmp/GMfifo2421802" /scratch/union/staging_dir/host/bin/ninja   -C /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24 
ninja: Entering directory `/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24'
[4/26] Building C object vcgencmd/CMakeFiles/vcgencmd.dir/vcgencmd.c.o
In function 'gencmd',
    inlined from 'main' at /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/vcgencmd/vcgencmd.c:152:14:
/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/vcgencmd/vcgencmd.c:109:4: warning: 'strncat' specified bound 1024 equals destination size [-Wstringop-overflow=]
  109 |    strncat(result, (const char *)(p+6), result_len);
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[9/26] Building C object vclog/CMakeFiles/vclog.dir/vclog.c.o
FAILED: vclog/CMakeFiles/vclog.dir/vclog.c.o 
/scratch/union/staging_dir/toolchain-aarch64_cortex-a76_gcc-14.2.0_musl/bin/aarch64-openwrt-linux-musl-gcc   -mcpu=cortex-a76 -O3 -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24=bcm27xx-utils-2024.04.24 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wall -Wextra -Werror -pedantic -DNDEBUG -MD -MT vclog/CMakeFiles/vclog.dir/vclog.c.o -MF vclog/CMakeFiles/vclog.dir/vclog.c.o.d -o vclog/CMakeFiles/vclog.dir/vclog.c.o -c /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/vclog/vclog.c
/scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/vclog/vclog.c: In function 'main':
[... cut ...]
make[2]: *** [Makefile:66: /scratch/union/build_dir/target-aarch64_cortex-a76_musl/bcm27xx-utils-2024.04.24/.built] Error 1
make[2]: Leaving directory '/scratch/union/package/utils/bcm27xx-utils'
time: package/utils/bcm27xx-utils/compile#2.45#0.26#0.91
    ERROR: package/utils/bcm27xx-utils failed to build.
make[1]: *** [package/Makefile:179: package/utils/bcm27xx-utils/compile] Error 1
make[1]: Leaving directory '/scratch/union'
make: *** [/scratch/union/include/toplevel.mk:248: package/bcm27xx-utils/compile] Error 2
```

pelwell, 2024-10-29:
Are you sure you're fully updated? That compilation error should have been fixed by #86.

graysky2, 2024-10-30:
Hi Phil - not sure what happened with the buildroot I used yesterday.  When I repeated this today, building the same commit, it did indeed complete successfully.  Sorry for the confusion.

closed 2024-10-30 by graysky2

pelwell, 2024-10-30:
It's nice when problems just go away all by themselves.