raspberrypi/utils #103: dtmerge, dtoverlay seem to require libdtovl.so from raspberrypi/userland

srcshelton, 2024-11-22:
https://github.com/raspberrypi/userland says that it is "ancient and deprecated"

However, `/usr/bin/dtmerge` and `/usr/bin/dtoverlay` from this repo still seem to link to `libdtovl.so` which only seems to be provided by that repo - and building this repo succeeds despite the library not being available:

```
# ldd /usr/bin/dtmerge
        linux-vdso.so.1 (0x00007ffec4484000)
        libdtovl.so => not found
        libc.so.6 => /lib64/libc.so.6 (0x00007ffec4250000)
        /lib/ld-linux-aarch64.so.1 => /lib64/ld-linux-aarch64.so.1 (0x00007ffec444c000)

# ldd /usr/bin/dtoverlay
        linux-vdso.so.1 (0x00007fff5c140000)
        libdtovl.so => not found
        libfdt.so.1 => /usr/lib64/libfdt.so.1 (0x00007fff5c0b0000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fff5bee0000)
        /lib/ld-linux-aarch64.so.1 => /lib64/ld-linux-aarch64.so.1 (0x00007fff5c108000)

# /usr/bin/dtmerge --help
/usr/bin/dtmerge: error while loading shared libraries: libdtovl.so: cannot open shared object file: No such file or directory
```

pelwell, 2024-11-22:
That's not my experience:
```
pi@raspberrypi:~/utils/dtmerge$ ldd dtoverlay
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v8l.so (0xf7ac0000)
        libfdt.so.1 => /lib/arm-linux-gnueabihf/libfdt.so.1 (0xf7a80000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xf7907000)
        /lib/ld-linux-armhf.so.3 (0xf7ae1000)
pi@raspberrypi:~/utils/dtmerge$ ldd dtmerge
        /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v8l.so (0xf78b0000)
        libfdt.so.1 => /lib/arm-linux-gnueabihf/libfdt.so.1 (0xf7870000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xf76f7000)
        /lib/ld-linux-armhf.so.3 (0xf78d3000)
```
Start with a clean build - don't bother to install it, just look at the built files.

srcshelton, 2024-11-22:
I'm not a CMake expert, but https://github.com/raspberrypi/utils/blob/422346057b7041468eb5c894ba0829bdbc6f4280/dtmerge/CMakeLists.txt#L14 does state:

```
dd_library (dtovl ${STATIC} dtoverlay.c)
target_link_libraries(dtovl fdt)
```

… and `dtoverlay` seems to be being built from the same directory - so I wonder if the difference between your build and mine is the value of `${STATIC}` causing your libdtovl to be statically linked but mine to be dynamic?

(The old version - https://github.com/raspberrypi/userland/blob/96a7334ae9d5fc9db7ac92e59852377df63f1848/helpers/dtoverlay/CMakeLists.txt#L20 - was explicitly linked as `${SHARED}`)

That said, I do see that `dtmerge/libdtovl.so` is built but not installed by this repo - so it seems likely that if `${STATIC}` is not set then the actual issue is that `libdtovl.so` from _this_ repo isn't being installed when it should be (or the linkage should always be static)?

pelwell, 2024-11-22:
I'm not setting any options or passing in any parameters. My build steps are:
```
$ cmake .
$ make
```
How about you?

pelwell, 2024-11-23:
I'll take suggestions as to how to improve the shared library case.

srcshelton, 2024-11-24:
Build log:

```
cmake -C /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280_build/gentoo_toolchain.cmake /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280
loading initial cache file /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280_build/gentoo_common_config.cmake
-- The C compiler identification is GNU 13.3.1
-- The CXX compiler identification is GNU 13.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/aarch64-unknown-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/aarch64-unknown-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- <<< Gentoo configuration >>>
Build type      RelWithDebInfo
Install path    /usr
Compiler flags:
C               -mcpu=cortex-a72+crc -O2 -pipe
C++             -mcpu=cortex-a72+crc -O2 -pipe
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed
Module          -Wl,-O1 -Wl,--as-needed
Shared          -Wl,-O1 -Wl,--as-needed

-- Configuring done (1.6s)
-- Generating done (0.1s)
-- Build files have been written to: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280_build
>>> Source configured.
>>> Compiling source in /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280 ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280_build"
ninja -v -j3 -l0
[1/42] /usr/bin/aarch64-unknown-linux-gnu-gcc    -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror   -ffunction-sections -MD -MT dtmerge/CMakeFiles/dtmerge.dir/dtmerge.c.o -MF dtmerge/CMakeFiles/dtmerge.dir/dtmerge.c.o.d -o dtmerge/CMakeFiles/dtmerge.dir/dtmerge.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/dtmerge/dtmerge.c
[2/42] /usr/bin/aarch64-unknown-linux-gnu-gcc    -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror   -ffunction-sections -MD -MT dtmerge/CMakeFiles/dtoverlay.dir/utils.c.o -MF dtmerge/CMakeFiles/dtoverlay.dir/utils.c.o.d -o dtmerge/CMakeFiles/dtoverlay.dir/utils.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/dtmerge/utils.c
[3/42] /usr/bin/aarch64-unknown-linux-gnu-gcc    -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror   -ffunction-sections -MD -MT dtmerge/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o -MF dtmerge/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o.d -o dtmerge/CMakeFiles/dtoverlay.dir/dtoverlay_main.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/dtmerge/dtoverlay_main.c
[4/42] /usr/bin/aarch64-unknown-linux-gnu-gcc    -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror   -ffunction-sections -MD -MT eeptools/CMakeFiles/eepmake.dir/eeplib.c.o -MF eeptools/CMakeFiles/eepmake.dir/eeplib.c.o.d -o eeptools/CMakeFiles/eepmake.dir/eeplib.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/eeptools/eeplib.c
[5/42] /usr/bin/aarch64-unknown-linux-gnu-gcc    -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror   -ffunction-sections -MD -MT eeptools/CMakeFiles/eepmake.dir/eepmake.c.o -MF eeptools/CMakeFiles/eepmake.dir/eepmake.c.o.d -o eeptools/CMakeFiles/eepmake.dir/eepmake.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/eeptools/eepmake.c
[6/42] /usr/bin/aarch64-unknown-linux-gnu-gcc    -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror   -ffunction-sections -MD -MT eeptools/CMakeFiles/eepdump.dir/eepdump.c.o -MF eeptools/CMakeFiles/eepdump.dir/eepdump.c.o.d -o eeptools/CMakeFiles/eepdump.dir/eepdump.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/eeptools/eepdump.c
[7/42] /usr/bin/aarch64-unknown-linux-gnu-gcc    -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror   -ffunction-sections -MD -MT eeptools/CMakeFiles/eepdump.dir/eeplib.c.o -MF eeptools/CMakeFiles/eepdump.dir/eeplib.c.o.d -o eeptools/CMakeFiles/eepdump.dir/eeplib.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/eeptools/eeplib.c
[8/42] /usr/bin/aarch64-unknown-linux-gnu-gcc -Ddtovl_EXPORTS   -mcpu=cortex-a72+crc -O2 -pipe -Wall -Wextra -Werror -fPIC   -ffunction-sections -MD -MT dtmerge/CMakeFiles/dtovl.dir/dtoverlay.c.o -MF dtmerge/CMakeFiles/dtovl.dir/dtoverlay.c.o.d -o dtmerge/CMakeFiles/dtovl.dir/dtoverlay.c.o -c /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/work/utils-422346057b7041468eb5c894ba0829bdbc6f4280/dtmerge/dtoverlay.c
[... cut ...]
set(CMAKE_ASM-ATT_LINK_FLAGS "-nostdlib" CACHE STRING "ASM-ATT link flags" FORCE)
set(CMAKE_C_COMPILE_OBJECT "<CMAKE_C_COMPILER> <DEFINES> <INCLUDES>  <FLAGS> -o <OBJECT> -c <SOURCE>" CACHE STRING "C compile command" FORCE)
set(CMAKE_CXX_COMPILE_OBJECT "<CMAKE_CXX_COMPILER> <DEFINES> <INCLUDES>  <FLAGS> -o <OBJECT> -c <SOURCE>" CACHE STRING "C++ compile command" FORCE)
set(CMAKE_Fortran_COMPILE_OBJECT "<CMAKE_Fortran_COMPILER> <DEFINES> <INCLUDES> -mcpu=cortex-a72+crc -O2 -pipe <FLAGS> -o <OBJECT> -c <SOURCE>" CACHE STRING "Fortran compile command" FORCE)
```

srcshelton, 2024-11-24:
Building with make instead of ninja:

`CMakeFiles/Makefile.cmake`
```
# CMAKE generated file: DO NOT EDIT!
# Generated by "Unix Makefiles" Generator, CMake Version 3.30

# The generator used is:
set(CMAKE_DEPENDS_GENERATOR "Unix Makefiles")

# The top level Makefile was generated from the following files:
set(CMAKE_MAKEFILE_DEPENDS
  "CMakeCache.txt"
  "/usr/share/cmake/Modules/CMakeCCompiler.cmake.in"
  "/usr/share/cmake/Modules/CMakeCCompilerABI.c"
  "/usr/share/cmake/Modules/CMakeCInformation.cmake"
  "/usr/share/cmake/Modules/CMakeCXXCompiler.cmake.in"
  "/usr/share/cmake/Modules/CMakeCXXCompilerABI.cpp"
  "/usr/share/cmake/Modules/CMakeCXXInformation.cmake"
  "/usr/share/cmake/Modules/CMakeCommonLanguageInclude.cmake"
  "/usr/share/cmake/Modules/CMakeCompilerIdDetection.cmake"
  "/usr/share/cmake/Modules/CMakeDetermineCCompiler.cmake"
  "/usr/share/cmake/Modules/CMakeDetermineCXXCompiler.cmake"
  "/usr/share/cmake/Modules/CMakeDetermineCompiler.cmake"
  "/usr/share/cmake/Modules/CMakeDetermineCompilerABI.cmake"
  "/usr/share/cmake/Modules/CMakeDetermineCompilerId.cmake"
  "/usr/share/cmake/Modules/CMakeDetermineCompilerSupport.cmake"
  "/usr/share/cmake/Modules/CMakeDetermineSystem.cmake"
  "/usr/share/cmake/Modules/CMakeFindBinUtils.cmake"
  "/usr/share/cmake/Modules/CMakeGenericSystem.cmake"
  "/usr/share/cmake/Modules/CMakeInitializeConfigs.cmake"
  "/usr/share/cmake/Modules/CMakeLanguageInformation.cmake"
  "/usr/share/cmake/Modules/CMakeParseImplicitIncludeInfo.cmake"
  "/usr/share/cmake/Modules/CMakeParseImplicitLinkInfo.cmake"
  "/usr/share/cmake/Modules/CMakeParseLibraryArchitecture.cmake"
  "/usr/share/cmake/Modules/CMakeSystem.cmake.in"
  "/usr/share/cmake/Modules/CMakeSystemSpecificInformation.cmake"
  "/usr/share/cmake/Modules/CMakeSystemSpecificInitialize.cmake"
  "/usr/share/cmake/Modules/CMakeTestCCompiler.cmake"
  "/usr/share/cmake/Modules/CMakeTestCXXCompiler.cmake"
  "/usr/share/cmake/Modules/CMakeTestCompilerCommon.cmake"
  "/usr/share/cmake/Modules/CMakeUnixFindMake.cmake"
  "/usr/share/cmake/Modules/Compiler/ADSP-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/ARMCC-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/ARMClang-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/AppleClang-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/Borland-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/Bruce-C-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/CMakeCommonCompilerMacros.cmake"
  "/usr/share/cmake/Modules/Compiler/Clang-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/Clang-DetermineCompilerInternal.cmake"
  "/usr/share/cmake/Modules/Compiler/Compaq-C-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/Compaq-CXX-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/Cray-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/CrayClang-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/Embarcadero-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/Fujitsu-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/FujitsuClang-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/GHS-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/GNU-C-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/GNU-C.cmake"
  "/usr/share/cmake/Modules/Compiler/GNU-CXX-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/GNU-CXX.cmake"
  "/usr/share/cmake/Modules/Compiler/GNU-FindBinUtils.cmake"
  "/usr/share/cmake/Modules/Compiler/GNU.cmake"
  "/usr/share/cmake/Modules/Compiler/HP-C-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/HP-CXX-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/IAR-DetermineCompiler.cmake"
  "/usr/share/cmake/Modules/Compiler/IBMCPP-C-DetermineVersionInternal.cmake"
  "/usr/share/cmake/Modules/Compiler/IBMCPP-CXX-DetermineVersionInternal.cmake"
[... 592 lines cut ...]
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/kdtc
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/otpset
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/ovmerge
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/pinctrl
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/share/bash-completion/completions/pinctrl
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/apitest
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/piotest
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/piopwm
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/pioseq
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/piows2812
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/dpi_interlace
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/rp1sm
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/bin/vcgencmd
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/share/man/man1/vcgencmd.1
-- Installing: /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image/usr/share/bash-completion/completions/vcgencmd
[... cut ...]
>>> Completed installing sys-apps/raspberrypi-tools-1.20241121 into /var/tmp/portage/sys-apps/raspberrypi-tools-1.20241121/image
```

Same result as before - `dtmerge` and `dtoverlay` are dynamically linked to `libdtovl.so`, which isn't being installed.  As mentioned, I'm relatively new to cmake, and I can't see any mention of STATIC or anything which would cause it to be set (unless something along the lines of performing a Release with Debug Info build automagically sets this?) to I'm stumped :(

srcshelton, 2024-11-24:
Hmm - `CMakeCache.txt` does contain `BUILD_SHARED_LIBS:BOOL=ON`…

Nope, setting `BUILD_SHARED_LIBS=OFF` and changing `${STATIC}` to `STATIC` still generates dynamically-linked binaries.

pelwell, 2024-11-24:
What happens if you change all occurrences of dtovl in this CMakeLists.txt to something else?

srcshelton, 2024-11-25:
Before:

```
# grep -R dtovl .
./dtmerge/CMakeLists.txt:add_library (dtovl ${STATIC} dtoverlay.c)
./dtmerge/CMakeLists.txt:target_link_libraries(dtovl fdt)
./dtmerge/CMakeLists.txt:target_link_libraries(dtmerge dtovl)
./dtmerge/CMakeLists.txt:target_link_libraries(dtoverlay dtovl)
```

After:

```
# grep -R dtovl .
./dtmerge/CMakeLists.txt:add_library (_not_dtovl ${STATIC} dtoverlay.c)
./dtmerge/CMakeLists.txt:target_link_libraries(_not_dtovl fdt)
./dtmerge/CMakeLists.txt:target_link_libraries(dtmerge _not_dtovl)
./dtmerge/CMakeLists.txt:target_link_libraries(dtoverlay _not_dtovl)
```

The source builds and installs without error, but:

```
# ldd ../image/usr/bin/dtmerge 
        linux-vdso.so.1 (0x0000007fae2a2000)
        lib_not_dtovl.so => not found
        libc.so.6 => /lib64/libc.so.6 (0x0000007fae090000)
        /lib/ld-linux-aarch64.so.1 => /lib64/ld-linux-aarch64.so.1 (0x0000007fae265000)
```

pelwell, 2024-11-25:
That add_library looks wrong - does replacing `${STATIC}` with `STATIC` improve (or even change) things for you? I seem to get the same Makefile either way, but perhaps the defaults are different on Gentoo.

srcshelton, 2024-11-26:
Ah - changing `${STATIC}` to `STATIC` seems to get the same result you have, where `libdtovl.so` isn't installed but `dtmerge` and `dtoverlay` no longer link against it either (… and the resulting binaries are only 32 bytes larger than when dynamically linked!)

pelwell, 2024-11-26:
Fixed by aaeac09.