raspberrypi/utils #117: piolib: Please add some method of unblocking DMA write

henryjliu, 2025-01-24:
All I want to do is stream data without gaps but it doesn't seem possible on RP1 pio.  

I set a large buffer:
#define BUF_SIZE 32768
uint8_t databuf[BUF_SIZE];
pio_sm_config_xfer(pio, sm, PIO_DIR_TO_SM, BUF_SIZE, 1);
    while (1) {        
        pio_sm_xfer_data(pio, sm, PIO_DIR_TO_SM, sizeof(databuf), databuf);
        sleep_ms(1); //simulate some calculation here        
    }

This works fine but the sleep_ms(1) inserts 1ms delay even though packet is 42ms long.

Doing:     
while (1) {        
        pio_sm_xfer_data(pio, sm, PIO_DIR_TO_SM, sizeof(databuf), databuf);
    }

In attempts to do continous data out, adds random blips of 100us and up to 5ms spacing.

Is there anyway around this eg automatic ring buffer or ping pong buffer?  I don't understand what the last argument in pio_sm_config_xfer(pio, sm, PIO_DIR_TO_SM, BUF_SIZE, 1); does: different values doesn't seem to do anything -> any clues?

pelwell, 2025-01-24:
Did you read the piolib header file? https://github.com/raspberrypi/utils/blob/master/piolib/include/piolib.h#L273 says:
```
static inline int pio_sm_config_xfer(PIO pio, uint sm, uint dir, uint buf_size, uint buf_count)
```
The fifth parameter is the number of DMA buffers to allocate - 1 is the worst possible value. Having at least 2 should singificantly reduce (but not eliminate) the gap between buffers.

henryjliu, 2025-01-24:
I read the header but all it says is: static inline int pio_sm_config_xfer(PIO pio, uint sm, uint dir, uint buf_size, uint buf_count)
{
    check_pio_param(pio);
    return pio->chip->pio_sm_config_xfer(pio, sm, dir, buf_size, buf_count);
}

so it's not clear what "buf_count" means.  It does perform a lot better now but it seems due to the new kernel sudo rpi-updates rpi-6.6.y  as recommended by https://github.com/raspberrypi/utils/issues/107

Even with new kernel my 32bit doesn't work as I put a comment on that closed topic - any other hints?  I did the sudo rpi-updates rpi-6.6.y and checked latest piolib pull and verified the apitest.c runs but it doesn't seem to output anything at the end and it's also FROM_SM and not TO_SM