raspberrypi/utils #118: piolib: Please provide piolib-dev or similar Debian package for developing software that uses /dev/pio

jepler, 2025-01-25:
.. a file that would include the necessary files to develop /dev/pio-using software without vendoring or submoduling the code from here.

PS with my Adafruit hat on it'd be nice if the license was permissive like MIT, too.

pelwell, 2025-04-16:
Recent changes to the piolib project have made installation of the library and headers possible. A proper package is in the pipeline.

macmpi, 2025-04-18:
If Alpine Linux is an option, [package](https://pkgs.alpinelinux.org/packages?name=raspberrypi-utils*&branch=edge&repo=&arch=aarch64&origin=&flagged=&maintainer=) is now part of the collection on Edge branch.

Gebhartj, 2025-04-28:
@pelwell, I'm wondering about the licensing of the piolib files. My understanding is that while GPL-2.0 might be acceptable for header files, it seems to also apply to files containing implementation code.

I noticed [here](https://pkgs.alpinelinux.org/package/edge/community/aarch64/raspberrypi-utils-piolib-dev) that piolib-dev is listed under the BSD-3-Clause license.

Could you clarify if my understanding is correct that the GPL-2.0 license would prevent linking piolib into closed-source projects?

ladyada, 2025-04-29:
t0tes - @makermelissa when you're back this week plz add to your list :)